ori-drs/allan_variance_ros #2: About units of params Accel Random Walk and Rate Random Walk

narutojxl, 2021-11-18:
Hi author,
Thanks for your sharing so convient tool to community very much.
In code we average three axis `Accel Random Walk` to get `accelerometer_random_walk` of kalibr.  `accelerometer_random_walk`'s unit is `m/s^3/sqrt(HZ)`, namely `m/s^2/sqrt(s)`, but the `Accel Random Walk`'s unit is `m/s^2`, is there a typo ?  The same question exist about `Rate Random Walk`.  `Velocity Random Walk` and `Angle Random Walk` units are equal to `accelerometer_noise_density` and `gyroscope_noise_density`of kalibr.   Thanks for your help and time!
https://github.com/ori-drs/allan_variance_ros/blob/25a33882f1fafddc910dbbdb0e2d4858cde009fc/scripts/analysis.py#L121

raabuchanan, 2021-11-18:
Hello,
There is actually a typo in the Kalibr documentation as discussed in this issue: https://github.com/ethz-asl/kalibr/issues/354 and in this issue of GTSAM: https://github.com/borglab/gtsam/issues/213

So `accelerometer_random_walk` should actually be `m/s^2` and not `m/s^2/sqrt(s),`

I believe the other parameters have the same units as in the Kalibr documentation so if there is something not matching, could you be more specific? Thanks

narutojxl, 2021-11-18:
Hi author, thanks for help very very much. 
According to kalibr `gyroscope_random_walk` unit is  `rad/s^2/sqrt(Hz)`, namely `rad/s/sqrt(s)`.  Your `Rate Random Walk` unit is `rad/s`, so i mistake there is also a typo in `Rate Random Walk`, sorry for that.
Maybe kalibr units of `\sigma_{bg}` and `\sigma_{ba}` should be corrected by someone. My problem is solved :)

closed 2021-11-18 by narutojxl

mintar, 2021-11-26:
Actually, the Kalibr documentation is correct, and ethz-asl/kalibr#354 and borglab/gtsam#213 are wrong. I've already replied in those two issues.

@narutojxl wrote:

> `accelerometer_random_walk`'s unit is `m/s^3/sqrt(HZ)`, namely `m/s^2/sqrt(s)`, but the `Accel Random Walk`'s unit is `m/s^2`, is there a typo ? The same question exist about `Rate Random Walk`.

You are correct. The units should be `m/s^3/sqrt(s)` and `deg/s^2/sqrt(s)`. PR #4 fixes that.

narutojxl, 2021-11-27:
Hi @mintar, thanks for your detailed explaination. I read in [your PR4](https://github.com/ori-drs/allan_variance_ros/pull/4),  and [kalibr/issues/354 comment](https://github.com/ethz-asl/kalibr/issues/354#issuecomment-979934812).
sigma_bg unit is `(rad/s)/sqrt(s) or (deg/s)/sqrt(s)`, why it is `deg/s^2/sqrt(s)` ?
sigma_ba unit is `(m/s^2)/sqrt(s)`, why it is `m/s^3/sqrt(s)`?   
I'm a little confused about this and don't  understand it . Could you please give me some info, thanks a lot.
>You were almost correct. The units should be m/s^3/sqrt(s) and deg/s^2/sqrt(s) (or equivalent). PR #4 fixes that.

[image]

mintar, 2021-11-27:
@narutojxl : Great catch! Yes, you were right all the time. The tables I've posted are correct, but I made a mistake in my comment to you and in #4. I've made a pull request to fix this (#8).

narutojxl, 2021-11-29:
Thanks @mintar :).