ori-drs/allan_variance_ros #3: analysis.py:25: RuntimeWarning: divide by zero encountered in log

narutojxl, 2021-11-19:
Hello author,
 I have recorded  3hr 37:31s (13051s)  2021-11-18-14-38-03.bag, it only contains imu data which published by gazebo `libgazebo_ros_imu_sensor.so`. `cooked_rosbag_2021-11-18-14-38-03.bag` is located in ` /home/jxl/jxl/allan_ws/ `.
> 
	rosrun allan_variance_ros cookbag.py  --input /home/jxl/2021-11-18-14-38-03.bag  --output  cooked_rosbag_2021-11-18-14-38-03.bag﻿
	rosrun allan_variance_ros allan_variance  /home/jxl/jxl/allan_ws/ /home/jxl/jxl/allan_ws/src/allan_variance_ros/config/gazebo_husky.yaml

When createing `allan_variance.csv`, it reminds me many imu msgs timestamp have somewrong, this is only part of the information output by the terminal.

>
	[ERROR] [1637290175.093369303]: Skipped imu messages 614547
	[ INFO] [1637290175.093390054]: 12291 / 13050 seconds loaded
	[ERROR] [1637290175.093418470]: IMU message before last imu time. IMU time: 13077039999998 last imu time 93877317622759
	[ERROR] [1637290175.093438070]: Skipped imu messages 614548
	[ INFO] [1637290175.093457149]: 12291 / 13050 seconds loaded
	[ERROR] [1637290175.093483640]: IMU message before last imu time. IMU time: 13077059999998 last imu time 93877317622759
	[ERROR] [1637290175.093503903]: Skipped imu messages 614549
	[ERROR] [1637290175.093530664]: IMU message before last imu time. IMU time: 13077079999998 last imu time 93877317622759
	[ERROR] [1637290175.093551317]: Skipped imu messages 614550
	[ INFO] [1637290175.093572197]: 12291 / 13050 seconds loaded
	[ERROR] [1637290175.093600659]: IMU message before last imu time. IMU time: 13077099999998 last imu time 93877317622759

	[ INFO] [1637290176.968499108]: Computed 0 averages for period 985.5
	[ INFO] [1637290176.968519090]: Computed 0 averages for period 985.6
	[ INFO] [1637290176.968539519]: Computed 0 averages for period 985.7
	[ INFO] [1637290176.968559882]: Computed 0 averages for period 985.8
	[ INFO] [1637290176.968580278]: Computed 0 averages for period 985.9
	[ INFO] [1637290176.968600411]: Computed 0 averages for period 986
	[ INFO] [1637290176.968620740]: Computed 0 averages for period 986.1
	[ INFO] [1637290176.968641282]: Computed 0 averages for period 986.2
	[ INFO] [1637290176.968661284]: Computed 0 averages for period 986.3
	[ INFO] [1637290176.968681675]: Computed 0 averages for period 986.4
	[ INFO] [1637290176.968701987]: Computed 0 averages for period 986.5
	[ INFO] [1637290176.968721859]: Computed 0 averages for period 986.6
	[ INFO] [1637290176.968742431]: Computed 0 averages for period 986.7
	[ INFO] [1637290176.968763563]: Computed 0 averages for period 986.8
	[ INFO] [1637290176.968783902]: Computed 0 averages for period 986.9
	[ INFO] [1637290176.968804266]: Computed 0 averages for period 987
	[ INFO] [1637290176.968824155]: Computed 0 averages for period 987.1
	[ INFO] [1637290176.968844767]: Computed 0 averages for period 987.2
	[ INFO] [1637290176.968865573]: Computed 0 averages for period 987.3

When i `rosrun allan_variance_ros analysis.py --data /home/jxl/jxl/allan_ws/allan_variance.csv`, program is terminated.  Is there something i missing?  Every imu msg has a timestamp, does it mean `cookbag.py` do nothing actually ?  Thanks for any your suggestion and help. 
>
	➜  [/home/jxl/jxl/allan_ws] rosrun allan_variance_ros analysis.py --data  /home/jxl/allan_variance.csv 
	/home/jxl/jxl/allan_ws/src/allan_variance_ros/scripts/analysis.py:25: RuntimeWarning: divide by zero encountered in log
	  logy = np.log(y)
	Traceback (most recent call last):
	  File "/home/jxl/jxl/allan_ws/src/allan_variance_ros/scripts/analysis.py", line 84, in <module>
	    accel_wn_intercept_x, xfit_wn = get_intercept(period[0:white_noise_break_point], acceleration[0:white_noise_break_point,0], -0.5, 1.0)
	  File "/home/jxl/jxl/allan_ws/src/allan_variance_ros/scripts/analysis.py", line 26, in get_intercept
	    coeffs, _ = curve_fit(line_func, logx, logy, bounds=([m, -np.inf], [m + 0.001, np.inf]))
	  File "/home/jxl/.local/lib/python3.8/site-packages/scipy/optimize/minpack.py", line 735, in curve_fit
	    ydata = np.asarray_chkfinite(ydata, float)
	  File "/usr/lib/python3/dist-packages/numpy/lib/function_base.py", line 495, in asarray_chkfinite
	    raise ValueError(
	ValueError: array must not contain infs or NaNs
	➜  [/home/jxl/jxl/allan_ws] 
This is csv file's part info.
>
	0.5 -0 -0 -0 -0 -0 -0 
	0.6000000238418579102 -0 -0 -0 -0 -0 -0 
	0.6999999880790710449 -0 -0 -0 -0 -0 -0 
	0.8000000119209289551 -0 -0 -0 -0 -0 -0 
	0.8999999761581420898 -0 -0 -0 -0 -0 -0 
	1 -0 -0 -0 -0 -0 -0 
	1.10000002384185791 -0 -0 -0 -0 -0 -0 
	1.20000004768371582 -0 -0 -0 -0 -0 -0 
	1.29999995231628418 -0 -0 -0 -0 -0 -0 
	1.39999997615814209 -0 -0 -0 -0 -0 -0 
	1.5 -0 -0 -0 -0 -0 -0

raabuchanan, 2021-11-19:
Interesting, I haven't tested this with simulated data. It seems like something is inconsistent with the timestamps of the data and `allan_variance_ros` is rejecting all the data. I'll give it a try with some gazebo data myself.

raabuchanan, 2021-11-23:
Can you let me know if my PR resolves the issue?
Thanks

narutojxl, 2021-11-24:
Hi author,
It worked with `fix-timestamp-init` branch, thanks for your help. Close it now.

closed 2021-11-24 by narutojxl

boeun98, 2022-04-13:
I have a same problem.
How can I fix it?