ori-drs/allan_variance_ros #9: What is the meaning of measure_rate in the configuration file? Is it related to the IMU sampling rate?

mppcasc, 2021-12-02:
Thanks

seajayshore, 2022-03-16:
+1 
Can someone please confirm the difference between measure_rate & imu_rate in the config files?

feixh, 2022-04-06:
Did anyone figure out what measure_rate means?

stephanietsuei, 2022-05-19:
Looking through the code, this is what I think is going on...

`measure_rate` can be set to a number smaller than `imu_rate` to subsample the IMU data. This means that the total computation will take less time, but that you are throwing away higher-frequency information. In my own little experiment, I set `measure_rate` to 1/4 `imu_rate`. There was very little change to the values of `accelerometer_random_walk` and `gyroscope_random_walk`, but the noise parameters were between 50 and 100% larger when the data was subsampled.

raabuchanan, 2022-08-03:
Yes @stephanietsuei is correct. I've updated the comment in the config file to clarify this.

closed 2022-08-03 by raabuchanan