ori-drs/allan_variance_ros #11: Failed to load Python extension for LZ4 support. LZ4 compression will not be available

John-Mccarty, 2022-02-19:
thanks for sharing your work.
when I try to run  rosrun allan_variance_ros cookbag.py --input workspace/imu_calibration.bag --output workspace/cooked_rosbag.bag
I get this error :Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
I instaled lz4 and roslz4 package but it still occur.What is the mistake?

boeun98, 2022-04-13:
I have a same problem. 
How can I solve that......?

raabuchanan, 2022-08-01:
Have you installed `rosbag` ? Try `pip install rosbag`

Also, this step is not strictly required. You can use a non pre-processed rosbag.

lmz-whu, 2022-08-09:
ROS1 depends on python2.7, then your env is python3. 
I solve this by reinstalling roslz4 to python3.
1. delete roslz4 in `/opt/ros/xxx/lib/python2.7/site-packages`
2. downlaod roslz4 from `https://rospypi.github.io/simple`
3. pip install

closed 2023-07-28 by raabuchanan