ori-drs/allan_variance_ros #13: What's the difference with imu_utils

qiaozhijian, 2022-02-25:
Hello~ I used this project[https://github.com/gaowenliang/imu_utils] before. I would like to ask the difference between the two.

seajayshore, 2022-03-17:
I'm not one of the developers but there is a discussion thread ([here](https://discourse.ros.org/t/open-source-allan-variance-tool-for-rosbags/23136)) between two of them that may answer some of your questions. 

Ultimately both do similar things but perhaps the methods & convenient output in the project is better for many people - plus the fact this project compiled on my machine (ROS Noetic) with no problems!

raabuchanan, 2022-08-01:
Thanks @seajayshore 
Yes, I would say this tool is basically just a simpler implementation and more designed for the ROS robotics community. Also it has been developed with the Camera-IMU calibration tool Kalibr in mind.

closed 2022-08-01 by raabuchanan