ori-drs/allan_variance_ros #15: How long should this take to run?

seajayshore, 2022-03-16:
Not complaining - trying to plan - but approx. how long should the allan_variance computation process take for say ~3hr long data? 

I have a rosbag ~3hrs (11214s) long being processed. 
The message `Computed X averages for period Y` is progressing through 1-unit of "period" every ~3.5s real-time. 
If I take "period" to mean the seconds in the rosbag it looks like my computer will take >10hrs to process... Is just how it is or am I missing something?

Obviously this is dependant on my hardware. I'm on a Ryzen 3600X based machine 16GB if that helps.

seajayshore, 2022-03-16:
Solved! 
I bothered to read the code (shame on me being lazy).

closed 2022-03-16 by seajayshore

v-lopez, 2022-08-30:
If you build with Release it's much faster.

I processed a 8h bag in 215seconds on a 3950X.