ori-drs/allan_variance_ros #22: ValueError with cookbag.py and analysis.py

MagdaZal, 2022-08-12:
- Hi, I've recorded IMU data (all topics) with SBG Ellipse. When I turn on cookbag. py, I get this error:

```
> magda@magda-VirtualBox:~$ cd catkin_ws
> magda@magda-VirtualBox:~/catkin_ws$ source devel/setup.bash
> magda@magda-VirtualBox:~/catkin_ws$ rosrun allan_variance_ros cookbag.py --input /home/magda/catkin_ws/src/allan_variance_ros/2022-08-12-12-14-04.bag --output cooked_rosbag
> Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
> Traceback (most recent call last):
>   File "/home/magda/catkin_ws/src/allan_variance_ros/scripts/cookbag.py", line 14, in <module>
>     with rosbag.Bag(args.output, 'w', compression='lz4') as outbag:
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 445, in __init__
>     raise ValueError('compression must be one of: %s' % ', '.join(allowed_compressions))  
> **ValueError: compression must be one of: none, bz2**
```

- But then I run normally 

`> rosrun allan_variance_ros allan_variance /home/magda/catkin_ws/src/allan_variance_ros/ /home/magda/catkin_ws/src/allan_variance_ros/config/ellipse.yaml`

-  this works great and I get allan_variance.csv. Unfortunatelly the next step doesn't work again. I see this error:
```
> magda@magda-VirtualBox:~/catkin_ws$ rosrun allan_variance_ros analysis.py --data /home/magda/catkin_ws/src/allan_variance_ros/allan_variance.csv
> 
> /home/magda/catkin_ws/src/allan_variance_ros/scripts/analysis.py:23: RuntimeWarning: divide by zero encountered in log
>   logy = np.log(y)
> Traceback (most recent call last):
>   File "/home/magda/catkin_ws/src/allan_variance_ros/scripts/analysis.py", line 101, in <module>
>     accel_wn_intercept_x, xfit_wn = get_intercept(period[0:white_noise_break_point], acceleration[0:white_noise_break_point,0], -0.5, 1.0)
>   File "/home/magda/catkin_ws/src/allan_variance_ros/scripts/analysis.py", line 24, in get_intercept
>     coeffs, _ = curve_fit(line_func, logx, logy, bounds=([m, -np.inf], [m + 0.001, np.inf]))
>   File "/home/magda/.local/lib/python3.6/site-packages/scipy/optimize/minpack.py", line 734, in curve_fit
>     ydata = np.asarray_chkfinite(ydata, float)
>   File "/home/magda/.local/lib/python3.6/site-packages/numpy/lib/function_base.py", line 486, in asarray_chkfinite
>     "array must not contain infs or NaNs")
> **ValueError: array must not contain infs or NaNs**
```

- Additionally I have created a launch file in which I show IMU topic and frequency. In this topic IMU status, sensors values and computed orientation using Quaternion are listed.

Do you know where is the problem?
Thank you very much for the answer

mahammadirfan, 2022-08-18:
Hi,
I am getting the same error. Have you found any solution ?

THanks!

MagdaZal, 2022-08-19:
@mahammadirfan Unfortunately not, I am still looking for the answer

MagdaZal, 2022-08-19:
@mahammadirfan Write in terminal  `chmod gu+x cookbag.py` and then path to cookbag `rosrun allan_variance_ros cookbag.py --input original_rosbag --output cooked_rosbag`.  It solves the first problem but in my case the .csv file still has values of 0 and Nans and I dont know why

mahammadirfan, 2022-08-19:
Hey,
its working here, make sure you have right IMU rostopic, IMU update rate in your yml file, analysis.py file. I edited this and got the results.

MagdaZal, 2022-08-22:
Hi @mahammadirfan, Thank you. I changed those files and it worked.
I close this issue

radhasaraf, 2024-08-14:
It's missing in the README, but analysis.py also takes a 'config' argument to which you can provide the config file (with the imu topic and rate). Then you don't have to change it manually.