ori-drs/allan_variance_ros #25: results interpret

zoldaten, 2022-10-03:
hi!
i got calibration papers:
[image]
[image]

this is cheap mpu-650 (GY-521).
how to understand is it good or not ?
what "stars", "circles", "triangles" mean ?

raabuchanan, 2022-10-18:
Hi @zoldaten 
I think something is going wrong in generating the Allan Deviation plot. There should not be such large spike after 100s sampling time. Can you confirm all the parameters in the config file are correct? Also, did you re-organize the bag using 
`rosrun allan_variance_ros cookbag.py --input original_rosbag --output cooked_rosbag` ?

raabuchanan, 2022-10-18:
The circles indicate the T=1s intercept for white noise
The triangles are the lowest point which is the bias instability
The starts are the T=3s intercept for random walk

zoldaten, 2022-10-19:
hi @raabuchanan 
1.here is my config:
```
imu_topic: "/imu/data"
imu_rate: 200
measure_rate: 200 # Rate to which imu data is subsampled
sequence_time: 10800 # 3 hours in seconds
```
2. i did  re-organize using cooked_rosbag.

i will try with another mpu.

silviazpi, 2022-11-08:
Hey!
First of everything thank you very much for your work.
Secondly, Im trying to understand the results and I dont get how the conversion from rad/s to rad/s/sqrt(hz) is done to get white noise of gyro and accel.
Thirdly, are 3h of data REALLY necessary? And also, do they have to be in a really quiet environment?
Also I wonder if this could be run in ros2 appart from the rosbag format.
Thanks!

raabuchanan, 2022-11-08:
Hi @silviazpi 
The units come from how Allan Deviation is computed. You can read about it in [this](https://upcommons.upc.edu/bitstream/handle/2117/103849/MScLeslieB.pdf?sequence=1&isAllowed=y) paper or directly from the [IEEE standard](https://ieeexplore.ieee.org/document/660628)

raabuchanan, 2022-11-08:
And yes, 3 hours is the minimum for good accuracy. If you look at the x axis of the example plot, we need up to 10^3 seconds sampling time. So 3 hours gives about 10 samples.

silviazpi, 2022-11-09:
@raabuchanan thank you very much for your quick response. The paper was really useful! Secondly, I found out that the /imu data can be published and recorded with ROS2 as long as you convert it afterwards using this library (really easy to use) https://gitlab.com/ternaris/rosbags. I thought that you might be interested in adding it to your README. Thanks again!