ori-drs/allan_variance_ros #28: Simulation Accuracy

goldbattle, 2022-10-19:
I am getting the following on the example simulation:
```
#Accelerometer
accelerometer_noise_density: 0.0025019929573561175 #true
accelerometer_noise_density: 0.0017989768146338795 

accelerometer_random_walk: 6.972435158192731e-05 #true
accelerometer_random_walk: 5.073249247970349e-05 

#Gyroscope
gyroscope_noise_density: 0.0001888339269965301  #true
gyroscope_noise_density: 0.00013494835013762752 

gyroscope_random_walk: 2.5565313322052523e-06 #true
gyroscope_random_walk: 2.324462789522549e-06 

rostopic: '/sensors/imu' 
update_rate: 400 
```

The `accelerometer_noise_density` seems to be almost half what it should be here.
[image]

Here is the script I am using to generate this result: [simulate_imu.sh.txt](https://github.com/ori-drs/allan_variance_ros/files/9821923/simulate_imu.sh.txt)

goldbattle, 2022-10-19:
Might likely be related to this change? https://github.com/ori-drs/allan_variance_ros/commit/7a3a461745ddfdd5b624aa904da8214bc8045f56#diff-6640893bc8a0db4aeabefce3f8108d038202212730673507002c46034698fa91L136-R233

goldbattle, 2022-10-19:
In reverting #23 I still get the same halved `accelerometer_noise_density`:
```
#Accelerometer
accelerometer_noise_density: 0.0025019929573561175 # true
accelerometer_noise_density: 0.0017989768146338795 
accelerometer_random_walk: 6.972435158192731e-05 # true
accelerometer_random_walk: 5.073249247970349e-05 

#Gyroscope
gyroscope_noise_density: 0.0001888339269965301 # true
gyroscope_noise_density: 0.00013494835013762752 
gyroscope_random_walk: 2.5565313322052523e-06 # true
gyroscope_random_walk: 2.324462789522549e-06 
```

In reverting https://github.com/ori-drs/allan_variance_ros/commit/7a3a461745ddfdd5b624aa904da8214bc8045f56 I get *good* results again:
```
#Accelerometer
accelerometer_noise_density: 0.0025019929573561175 #true
accelerometer_noise_density: 0.0025186945257589842 
accelerometer_random_walk: 6.972435158192731e-05 #true
accelerometer_random_walk: 7.940436509327152e-05 

#Gyroscope
gyroscope_noise_density: 0.0001888339269965301  #true
gyroscope_noise_density: 0.00018886224457709532 
gyroscope_random_walk: 2.5565313322052523e-06 #true
gyroscope_random_walk: 3.4042438707395636e-06 
```

raabuchanan, 2022-10-28:
Yes, there were two changes in that PR both of which are affecting the result. 
One was to use overlapping bins which I now think has a bug. The other change was rather than average the 3 axes, to be more conservative and use largest value.

If I set overlap to zero and use the average I get exactly the same as your second set of numbers. If I set overlap to zero and use the largest axis value I get slightly different numbers but still close:
```
#Accelerometer
accelerometer_noise_density: 0.0025019929573561175 #true
accelerometer_noise_density: 0.002561528431294409 
accelerometer_random_walk: 6.972435158192731e-05 #true
accelerometer_random_walk: 9.079971439106259e-05

#Gyroscope
gyroscope_noise_density: 0.0001888339269965301  #true
gyroscope_noise_density: 0.00019087356904881286 
gyroscope_random_walk: 2.5565313322052523e-06 #true
gyroscope_random_walk: 4.118240612064778e-06 
```

raabuchanan, 2022-10-28:
I will make a fix now to disable the overlap bug and in the future I'll add a parameter to use average axis vs worst axis. In my opinion though I find it's best to be more conservative with these parameters. 
Kalibr also recommends adding an order of magnitude which I find excessive https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#kalibr-imu-noise-parameters-in-practice

closed 2022-10-28 by raabuchanan

goldbattle, 2022-10-30:
Yeah, I think it is very reasonable to use the worst axis as the output as this is the most conservative. The issue is that with the overlap, the computed one was actually *more confident* then the true value, which is definitely is not something we want.

PR #31 (2022-10-28): Disable overlap and fix typo
Fixes #28