ori-drs/allan_variance_ros #35: IMU Data Collection

SunilSrivatsav9, 2023-06-15:
Hello. I am a newbie to IMU.
Can you help me out in collecting data from IMU using Rosbag?

Do we need to collect data by exciting all IMU axes (rotation and translation)?

In the description it is given collect data for 3 hrs on damped surface. Can you explain in detail?

raabuchanan, 2023-07-28:
Hi,
The IMU must be totally static for the whole 3 hours. By damped surface I mean something like foam padding. This is in case there are any vibrations, for example a nearby machine or person walking by. I recorded data by placing the IMU on foam padding in a remote corner of the lab away from anything that might cause vibration.

raabuchanan, 2023-09-08:
You can collect the data anywhere so long as the IMU is not moving. So if it's in a robot you should probably make sure all motors or large fans are turned off to reduce vibrations.

Iron/soft iron may make a tiny difference but since you'll be using the IMU inside the robot anyway it probably makes most sense to calibrate inside the robot.

closed 2023-09-08 by raabuchanan