ori-drs/allan_variance_ros #36: The diagram does not match the result

wbl17792615320, 2023-06-19:
Hello, in allan _ variance _ ros, why the data corresponding to the acceleration **circle and the star** can match the output result, while the data corresponding to the angular velocity circle and the star cannot match the output result, even several orders of magnitude worse.

**My results：**
[image]

[image]

**My results：**
#Accelerometer
accelerometer_noise_density: 0.0008376248418885106 
accelerometer_random_walk: 2.591042775935633e-05 

#Gyroscope
gyroscope_noise_density: 1.0752236145252607e-05 
gyroscope_random_walk: 6.9924971907009525e-09 

rostopic: '/sensors/imu' #Make sure this is correct
update_rate: 400.0 #Make sure this is correct

**ps:I modified the name and frequency of IMU topics in the yaml file, but obviously nothing changed in the results**

raabuchanan, 2023-07-28:
Hi, 
Could you please clarify what data you are using? Is it simulated or the rosbag I provided or your own data?

raabuchanan, 2023-07-28:
I just tried again, simulating a rosbag then running estimation on the rosbag and got the following:
```
#Accelerometer
accelerometer_noise_density: 0.002561528431294409 
accelerometer_random_walk: 9.079971439106259e-05 

#Gyroscope
gyroscope_noise_density: 0.00019087356904881286 
gyroscope_random_walk: 4.118240612064778e-06 

rostopic: '/sensors/imu' #Make sure this is correct
update_rate: 400.0 #Make sure this is correct
```

[image]

[image]

Did you use the correct config file in `allan_variance_ros/config/sim.yaml` ?