ori-drs/allan_variance_ros #43: Getting ValueError: array must not contain infs or NaNs while running analysis.py

RajS999, 2023-12-30:
I am trying to calculate accel and gyro noise and random walk for Intel realsense. I quickly generated imu csv in the format specified [here](https://github.com/ethz-asl/kalibr/wiki/bag-format) using custom CPP script for just 3 seconds. I wanted to check if it will work or not before recording 20 hours of IMU data. 

I created ros bag from csv by running:

```
rosrun kalibr kalibr_bagcreater --folder ./data/imu_data --output-bag ./data/newbag.bag
```
Checking the bag created:

```
root@e9408da6622e:/data# rosbag info /data/data/newbag.bag  
path:        /data/data/newbag.bag 
version:     2.0 
duration:    2.4s 
start:       Dec 30 2023 18:41:25.42 (1703961685.42) 
end:         Dec 30 2023 18:41:27.84 (1703961687.84) 
size:        312.7 KB 
messages:    830 
compression: none [1/1 chunks] 
types:       sensor_msgs/Imu [6a62c6daae103f4ff57a132d6f95cec2] 
topics:      /imu0   830 msgs    : sensor_msgs/Imu 
```

Then I converted it to allan variance compliant bag by running:

```
rosrun allan_variance_ros cookbag.py --input /data/data/imu_data/newbag.bag --output /data/data/imu_data/cooked_bag.bag 
```
Checking the bag created:

```
root@e9408da6622e:/catkin_ws# rosbag info /data/data/imu_data/cooked_bag.bag  
path:         /data/data/imu_data/cooked_bag.bag 
version:      2.0 
duration:     2.4s 
start:        Dec 30 2023 18:41:25.42 (1703961685.42) 
end:          Dec 30 2023 18:41:27.84 (1703961687.84) 
size:         41.7 KB 
messages:     830 
compression:  lz4 [1/1 chunks; 8.47%] 
uncompressed: 296.1 KB @ 122.5 KB/s 
compressed:    25.1 KB @  10.4 KB/s (8.47%) 
types:        sensor_msgs/Imu [6a62c6daae103f4ff57a132d6f95cec2] 
topics:       /imu0   830 msgs    : sensor_msgs/Imu
```

I created allan variance config file imu_confog.yaml:

```
imu_topic: "/imu0"
imu_rate: 346
measure_rate: 346 # since bag contains 830 messages / 2.4 seconds = 345.8333 approx 346
sequence_time: 3 # since bag contains 2.4 seconds worth of data
```
(I also tried with both rates 400, no luck.)

Here is how my data folder looks like:

[image]

Then I run allan variance:

```
root@e9408da6622e:/catkin_ws# rosrun allan_variance_ros allan_variance /data/data/imu_data /data/data/allan_variance/imu_config.yaml
```

And finally run Analyse.py:

```
root@e9408da6622e:/catkin_ws# rosrun allan_variance_ros analysis.py --data /data/data/imu_data/allan_variance.csv --config /data/data/allan_variance/imu_config.yaml
/catkin_ws/src/allan_variance_ros/scripts/analysis.py:23: RuntimeWarning: divide by zero encountered in log
  logy = np.log(y)
Traceback (most recent call last):
  File "/catkin_ws/src/allan_variance_ros/scripts/analysis.py", line 101, in <module>
    accel_wn_intercept_x, xfit_wn = get_intercept(period[0:white_noise_break_point], acceleration[0:white_noise_break_point,0], -0.5, 1.0)
  File "/catkin_ws/src/allan_variance_ros/scripts/analysis.py", line 24, in get_intercept
    coeffs, _ = curve_fit(line_func, logx, logy, bounds=([m, -np.inf], [m + 0.001, np.inf]))
  File "/usr/lib/python3/dist-packages/scipy/optimize/minpack.py", line 708, in curve_fit
    ydata = np.asarray_chkfinite(ydata, float)
  File "/usr/lib/python3/dist-packages/numpy/lib/function_base.py", line 495, in asarray_chkfinite
    raise ValueError(
ValueError: array must not contain infs or NaNs
```

But as you can see it is giving error `array must not contain infs or NaNs`. Turns out there are some NaNs in allan_variance.csv file generated. What I am doing wrong here?

I have uploaded all bags and csv file on drive [here](https://drive.google.com/drive/folders/15M2P3NHtgvJvRPZMNqd5uMvz1R0GN-0S?usp=sharing).

RajS999, 2024-01-03:
I guess capturing very tiny amount of data doesnt work. I tried capturing few seconds of data (for quickly checking if the whole pipeline will eventually work or not), even though doc says that we must record at least 1 hour of data, preferably at least 3 hours of data. I recorded data for 21 hours and it worked. Still I will like someone confirm that we should indeed get `ValueError: array must not contain infs or NaNs` error when working with small amount of data.

abolfazljalali, 2024-02-21:
Hi,

Could you solve this issue?

timengelbracht, 2025-08-03:
A bit late to the party, but in case anyone else struggles with this: I only recorded for ~ca. 45 min (on a GoPro 9, so 3 hours not really feasible, also u have to stitch recordings together, as GoPro splits recordings once they reach ~4Gb) and also got the "ValueError: array must not contain infs or NaNs" error when running analysis.py.
If u open the allan_variance.csv in text editor or similar u'll notice that at some point (for me around line 900) all lines will be filled with zeros and NaN up until the end. Just remove these lines and rerun analysis.py. Not the cleanest way obviously, but the script finishes without error :)

I'll leave this here incase anyone needs some rough esstimate gopro imu values:

ACCELEROMETER:
X Velocity Random Walk:  0.00122 m/s/sqrt(s)  0.07323 m/s/sqrt(hr)
Y Velocity Random Walk:  0.00111 m/s/sqrt(s)  0.06666 m/s/sqrt(hr)
Z Velocity Random Walk:  0.00132 m/s/sqrt(s)  0.07941 m/s/sqrt(hr)
X Bias Instability:  0.00020 m/s^2  2634.90278 m/hr^2
Y Bias Instability:  0.00037 m/s^2  4821.89112 m/hr^2
Z Bias Instability:  0.00039 m/s^2  5035.35269 m/hr^2
X Accel Random Walk:  0.00013 m/s^2/sqrt(s)
Y Accel Random Walk:  0.00043 m/s^2/sqrt(s)
Z Accel Random Walk:  0.00040 m/s^2/sqrt(s)
GYROSCOPE:
X Angle Random Walk:  0.01963 deg/sqrt(s)  1.17790 deg/sqrt(hr)
Y Angle Random Walk:  0.01278 deg/sqrt(s)  0.76697 deg/sqrt(hr)
Z Angle Random Walk:  0.00632 deg/sqrt(s)  0.37936 deg/sqrt(hr)
X Bias Instability:  0.00628 deg/s  22.59310 deg/hr
Y Bias Instability:  0.00000 deg/s  0.01773 deg/hr
Z Bias Instability:  0.00239 deg/s  8.58645 deg/hr
X Rate Random Walk:  0.00533 deg/s/sqrt(s)
Y Rate Random Walk:  0.00234 deg/s/sqrt(s)
Z Rate Random Walk:  0.00166 deg/s/sqrt(s)